OpenMind/OM1 #88: [Bounty 1] Please add support for building a local RAG system for user-provided PDFs.

paigexu522, 2025-02-12:
Guidelines:
• Develop a new input plugin to chunk PDFs and construct the knowledge base.
• Implement a new LLM plugin to query the knowledge base based on user prompts.

aybanda, 2025-02-12:
I’d like to give it a try if it’s open to everyone. Could you also provide the bounty amount you mentioned, if applicable?

paigexu522, 2025-02-13:
> I’d like to give it a try if it’s open to everyone. Could you also provide the bounty amount you mentioned, if applicable?

Hi - we already have someone worked on this issue. please check out other PR if interested. Bounty value coming up soon. you can try deploy locally first and let us know any questions!

closed 2025-03-30 by bobjiang