OpenMind/OM1 #110: [Bounty] Automate pyaudio dependency fix for Mac

kyle-openmind, 2025-02-18:
**Overview**
When installing OM1, `pyaudio` needs to be installed if `portaudio` is missing. We can automate this installation process by using a setup script.

**Features**

**Guidelines**
We don't have any setup script yet. We would need to start from scratch to create such a script. 

**License**
This project is licensed under the MIT License. See the [LICENSE](https://github.com/OpenmindAGI/OM1/blob/main/LICENSE) file for more details.

**Contact**
For questions or further assistance, reach out via the issue tracker or contact the maintainers directly.

TangmereCottage, 2025-03-13:
The way to deal with this is to make sure that people have `portaudio` installed, and then the `pyaudio` in the pyproject.toml will install cleanly in the `venv`. We could run a simple check for portaudio and ask them to install it if they do not have it.

TangmereCottage, 2025-03-20:
fixed

closed 2025-03-20 by TangmereCottage