OpenMind/OM1 #218: [BUG] Error opening video stream from /dev/video0

bobjiang, 2025-04-23:
**Describe the bug**
when run the OM1 by the command like
`uv run src/run.py my_config`
note: I have installed `ffmpeg` pkg.

I got the following error msg:
```
uv run src/run.py ainex
INFO:root:Checking for backup URID in your .env file.
WARNING:root:Could not find backup URID in your .env file. Using 'default'. Multirobot deployments will conflict.
INFO:root:No robot hardware ethernet port provided.
WARNING:root:Unitree SDK not found. Please install the Unitree SDK to use this plugin.
WARNING:root:Unitree SDK not found. Please install the Unitree SDK to use this plugin.
WARNING:root:Unitree SDK not found. Please install the Unitree SDK to use this plugin.
INFO:om1_utils:WebSocket client thread started
INFO:om1_vlm:Started video processing thread
[ WARN:0@32.396] global cap_v4l.cpp:913 open VIDEOIO(V4L2:/dev/video0): can't open camera by index
INFO:om1_utils:Registered message callback
[ERROR:0@32.402] global obsensor_uvc_stream_channel.cpp:158 getStreamChannelGroup Camera index out of range
INFO:om1_vlm:Using camera: /dev/video0
ERROR:om1_vlm:Error opening video stream from /dev/video0
INFO:om1_utils:Connected to wss://api-vila.openmind.org
INFO:om1_utils:Connection established
╭─────────────────────────────────────── Traceback (most recent call last) ────────────────────────────────────────╮
│ /home/ubuntu/OM1/src/run.py:22 in start                                                                          │
│                                                                                                                  │
│   19 │   runtime = CortexRuntime(config)                                                                         │
│   20 │                                                                                                           │
│   21 │   # Start the runtime                                                                                     │
│ ❱ 22 │   asyncio.run(runtime.run())                                                                              │
│   23                                                                                                             │
│   24                                                                                                             │
│   25 if __name__ == "__main__":                                                                                  │
│                                                                                                                  │
│ ╭─────────────────────────────────────────────────── locals ───────────────────────────────────────────────────╮ │
│ │      config = RuntimeConfig(                                                                                 │ │
│ │               │   hertz=10,                                                                                  │ │
│ │               │   name='AiNex-PersonDetection',                                                              │ │
│ │               │   system_prompt_base='You are an AiNex robot. Follow all instructions carefully and ensure   │ │
│ │               safe operat'+4,                                                                                │ │
│ │               │   system_governance='Execute actions only when sensor inputs are validated and all safety    │ │
│ │               protocols a'+7,                                                                                │ │
│ │               │   system_prompt_examples=[                                                                   │ │
│ │               │   │   '1. If the camera detects a person, wave your hand',                                   │ │
│ │               │   │   '2. When a person appears in view, perform the shake paw action.'                      │ │
│ │               │   ],                                                                                         │ │
│ │               │   agent_inputs=[<inputs.plugins.vlm_vila.VLMVila object at 0xffff9a9e6270>],                 │ │
│ │               │   cortex_llm=<llm.plugins.openai_llm.OpenAILLM object at 0xffff9b0717f0>,                    │ │
│ │               │   simulators=[],                                                                             │ │
│ │               │   agent_actions=[                                                                            │ │
│ │               │   │   AgentAction(                                                                           │ │
│ │               │   │   │   name='move',                                                                       │ │
│ │               │   │   │   llm_label='move',                                                                  │ │
│ │               │   │   │   interface=<class 'actions.move.interface.Move'>,                                   │ │
│ │               │   │   │   implementation=<actions.passthrough.PassthroughAction object at 0xffff336b62d0>,   │ │
│ │               │   │   │   connector=<actions.move.connector.ros2.MoveRos2Connector object at 0xffff2de4c9e0> │ │
│ │               │   │   )                                                                                      │ │
│ │               │   ],                                                                                         │ │
│ │               │   api_key='your_api_key_here',                                                               │ │
│ │               │   URID='default',                                                                            │ │
│ │               │   unitree_ethernet=None                                                                      │ │
│ │               )                                                                                              │ │
│ │ config_name = 'ainex'                                                                                        │ │
│ │       debug = False                                                                                          │ │
│ │     runtime = <runtime.cortex.CortexRuntime object at 0xffff99e54d10>                                        │ │
│ ╰──────────────────────────────────────────────────────────────────────────────────────────────────────────────╯ │
│                                                                                                                  │
│ /home/ubuntu/.local/share/uv/python/cpython-3.12.10-linux-aarch64-gnu/lib/python3.12/asyncio/runners.py:195 in   │
│ run                                                                                                              │
│                                                                                                                  │
[... 120 lines cut ...]
│ │ finished_promises = []                                                           │                             │
│ │     input_strings = [None]                                                       │                             │
│ │            inputs = [<inputs.plugins.vlm_vila.VLMVila object at 0xffff9a9e6270>] │                             │
│ │              self = <fuser.Fuser object at 0xffff99e54cb0>                       │                             │
│ ╰──────────────────────────────────────────────────────────────────────────────────╯                             │
╰──────────────────────────────────────────────────────────────────────────────────────────────────────────────────╯
TypeError: can only concatenate str (not "list") to str
```

**To Reproduce**
List the steps to replicate the bug:
1. uv run src/run.py my_config

**Expected behavior**
command should be successful

**Screenshots**
If relevant, attach screenshots to illustrate the problem.

**System Information (please complete the following information):**
 - **Operating System:** ubuntu 22.04
 - **Python Version:** python 3.8.10

**Additional context**
Include any extra details or related information that may help diagnose the issue.

bobjiang, 2025-04-23:
unplug and plug the camera,
video doesn't work,

tried to check the device working,

- **ffplay**
```
export DISPLAY=:0
ffplay /dev/video0

ffplay version 4.2.7-0ubuntu0.1 Copyright (c) 2003-2022 the FFmpeg developers
  built with gcc 9 (Ubuntu 9.4.0-1ubuntu1~20.04.1)
  configuration: --prefix=/usr --extra-version=0ubuntu0.1 --toolchain=hardened --libdir=/usr/lib/aarch64-linux-gnu --incdir=/usr/include/aarch64-linux-gnu --arch=arm64 --enable-gpl --disable-stripping --enable-avresample --disable-filter=resample --enable-avisynth --enable-gnutls --enable-ladspa --enable-libaom --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libcodec2 --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libjack --enable-libmp3lame --enable-libmysofa --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librsvg --enable-librubberband --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvidstab --enable-libvorbis --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx265 --enable-libxml2 --enable-libxvid --enable-libzmq --enable-libzvbi --enable-lv2 --enable-omx --enable-openal --enable-opencl --enable-opengl --enable-sdl2 --enable-libdc1394 --enable-libdrm --enable-libiec61883 --enable-chromaprint --enable-frei0r --enable-libx264 --enable-shared
  libavutil      56. 31.100 / 56. 31.100
  libavcodec     58. 54.100 / 58. 54.100
  libavformat    58. 29.100 / 58. 29.100
  libavdevice    58.  8.100 / 58.  8.100
  libavfilter     7. 57.100 /  7. 57.100
  libavresample   4.  0.  0 /  4.  0.  0
  libswscale      5.  5.100 /  5.  5.100
  libswresample   3.  5.100 /  3.  5.100
  libpostproc    55.  5.100 / 55.  5.100
Could not initialize SDL - No available video device
(Did you set the DISPLAY variable?)
```

- **mpv**
```
 v4l2-ctl --list-formats-ext -d /dev/video0
ioctl: VIDIOC_ENUM_FMT
	Type: Video Capture

	[0]: 'YUYV' (YUYV 4:2:2)
		Size: Discrete 640x480
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.040s (25.000 fps)
			Interval: Discrete 0.050s (20.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
			Interval: Discrete 0.100s (10.000 fps)
			Interval: Discrete 0.200s (5.000 fps)
❯ mpv av://v4l2:/dev/video0 --demuxer-lavf-o=video_size=640x480,input_format=mjpeg
[lavf] avformat_open_input() failed
Failed to recognize file format.

Exiting... (Errors when loading file)
```

TangmereCottage, 2025-05-08:
This is not a bug. You introduced erroneous characters in the `json5` config file, namely, `[]`, which breaks your config file. Note the TypeError: `TypeError: can only concatenate str (not "list") to str` which explains the problem.