OpenMind/OM1 #333: Add G1 state provider

openminddev, 2025-07-30:
Add G1 state provider

ChrisnaRiansyah, 2025-08-14:
Lfg

SatoshiZKMO, 2025-10-24:
Excited about this feature! We’re currently extending G1 modules with Health, Visual Feedback, and Voice Control. Integrating a state provider will allow these modules to share real-time robot status, improving monitoring and control.
Any ideas or suggestions for additional states or metrics we should include?

Prachi1615, 2025-12-30:
This has been implemented, closing the issue now.
Hi @SatoshiZKMO that sounds interesting. Let us know if you are still working on this. Happy to chat!

SatoshiZKMO, 2026-01-01:
Thanks for the update!

Hi @Prachi1615 

Yes, I’m still actively working on G1-related improvements.
I’ve opened a few PRs in this repo that might be relevant to this discussion:

• #12 – Add G1 Voice Control (VUIClient) starter module  
• #11 – Add G1 Health API and client  
• #7  – Fix RequestFutureQueue  

Happy to iterate, rebase, or adjust these based on your feedback.
Would be great to sync if this aligns with your current roadmap 

https://github.com/OpenMind/unitree_sdk2_python/pulls?q=is%3Aopen+is%3Apr

SatoshiZKMO, 2026-01-01:
Thanks for the update!

Hi @Prachi1615

Yes, I’m still actively working on G1-related improvements.
I’ve opened a few PRs in this repo that might be relevant to this
discussion:

• #12 – Add G1 Voice Control (VUIClient) starter module
• #11 – Add G1 Health API and client
• #7  – Fix RequestFutureQueue

Happy to iterate, rebase, or adjust these based on your feedback.
Would be great to sync if this aligns with your current roadmap

https://github.com/OpenMind/unitree_sdk2_python/pulls?q=is%3Aopen+is%3Apr

Prachi Sethi ***@***.***>, 30 Ara 2025 Sal, 23:06 tarihinde
şunu yazdı: