OpenMind/OM1 #354: [BUG] PyAudio installation fails: portaudio.h: No such file or directory on Ubuntu due to outdated python-all-dev in docs

ScapeXYZ, 2025-08-26:
**Describe the bug**
When following the documentation to install the PortAudio library for PyAudio, the command provided includes `python-all-dev`, which no longer exists on modern Ubuntu distributions. This causes PyAudio to fail to build with the error:

`src/_portaudiomodule.c:29:10: fatal error: portaudio.h: No such file or directory`

**To Reproduce**
Steps to replicate the bug:
1. Run the command from the docs on Ubuntu:
``sudo apt-get update
sudo apt-get install portaudio19-dev python-all-dev``
2. Attempt to install PyAudio with `pip install pyaudio`
3. Build fails with ``portaudio.h: No such file or directory``

**Expected behavior**
The provided install command should work on modern Ubuntu and allow PyAudio to compile successfully.

**Screenshots**
[image]

[image]
**Logs**
Error message during build:
``fatal error: portaudio.h: No such file or directory
compilation terminated.
error: command '/usr/bin/gcc' failed with exit code 1``

**System Information (please complete the following information):**
 - **Operating System:** [Ubuntu]
 - **Python Version:** [3.12]

**Additional context**
The issue is due to python-all-dev being deprecated and unavailable.
Installing **only** PortAudio development headers fixes the problem:

``sudo apt-get update
sudo apt-get install portaudio19-dev``

Updating the docs to remove python-all-dev would prevent this issue.

openminddev, 2025-08-26:
@ScapeXYZ Thanks for your context. We will update our docs.

Prachi1615, 2025-08-29:
Updated the docs

closed 2025-08-29 by Prachi1615