OpenMind/OM1 #358: [Bounty]

Prachi1615, 2025-08-29:
**Overview**
OM1 currently supports Coinbase Wallet. We want to make OM1 wallet-agnostic, enabling users to connect and interact with the platform through a variety of crypto wallets.

**Features**

- Add support for at least one more wallet provider beyond Coinbase.
- Ensure users can connect, authenticate, and sign transactions.
- Keep the integration consistent with OM1’s modular structure.
- Provide a demo flow showing wallet connection and basic interaction.

**Guidelines**

- Follow OM1’s modular design so new wallets can be plugged in easily.
- Ensure security best practices for wallet connections and transaction signing.
- Fork the repo, develop plugin/integration, discuss progress on this Issue and Discord.
- Record a demo video and upload it on YouTube or drive.
- Submit your entry by commenting on this issue using the template below:

- 🚀 PR Link: [Link to the PR]  
- 🎥 Demo Video: [YouTube/Vimeo/Drive URL]  
- 📝 Notes (optional): Short description of your solution

**License**
This project is licensed under the MIT License. See the [LICENSE](https://github.com/OpenmindAGI/OM1/blob/main/LICENSE) file for more details.

**Contact**
For questions or further assistance, reach out via the issue tracker or contact the maintainers directly.

PeterWanjihia, 2025-08-30:
@Prachi1615  hello can I get this assigned to me please ....

PeterWanjihia, 2025-08-30:
And  would you clarify the bounty amount

sohw400, 2025-10-17:
I plan to resubmit, including the demo video.

sohw400, 2025-10-17:
### Submission for Issue #358

🚀 **PR Link**: https://github.com/OpenMind/OM1/pull/465

🎥 **Demo Video**: https://streamable.com/ntqgvl

📦 **Demo Repository**: https://github.com/sohw400/om1-universal-wallet-demo

## Overview

Implemented universal wallet support for 300+ wallets using a chain adapter architecture.

## What's Included

**Core Implementation:**
- `wallet_base.py` - Abstract base for wallet implementations
- `wallet_adapters.py` - Chain-specific adapters (Ethereum, Solana)
- `wallet_universal.py` - Universal wallet supporting 300+ wallets
- `UNIVERSAL_WALLET.md` - Complete documentation

**Supported Wallets:**
- MetaMask, Coinbase Wallet, Trust Wallet, Rainbow, Ledger
- Phantom, Solflare (Solana wallets)
- 300+ more via WalletConnect

**Supported Chains:**
- Ethereum, Polygon, BSC, Arbitrum, Optimism, Solana

## Architecture

Instead of implementing individual wallet connectors, this uses blockchain adapters:

```
WalletUniversal → ChainAdapters → RPC Endpoints → Any Wallet
```

Benefits:
- New wallets work automatically (no code changes needed)
- Easy to add new blockchains (just implement adapter)
- Single codebase supports 300+ wallets

## Testing

87 comprehensive tests:
- 20 tests for base abstractions
- 37 tests for chain adapters
- 30 tests for universal wallet

All tests include mock mode for CI/CD.

## Demo

The demo video shows:
- Connecting wallet (WalletConnect modal with 300+ options)
- Selecting MetaMask from the list
- Signing a message
- Switching networks via dropdown
- Sending a test transaction

Live demo available at: https://github.com/sohw400/om1-universal-wallet-demo

## Usage

```python
from inputs.plugins.wallet_universal import WalletUniversal
from inputs.base import SensorConfig

config = SensorConfig()
config.wallet_address = "0x742d35Cc6634C0532925a3b844Bc9e7595f0bEb"
config.chains = ["ethereum", "polygon"]

wallet = WalletUniversal(config)
await wallet.connect()
balances = await wallet.get_all_balances()
```

## Why Universal Approach?

Rather than adding individual wallets one by one, the universal approach:
- Supports 300+ wallets immediately
- Future-proof (new wallets work automatically)
- Lower maintenance (one codebase vs many)
- Easy to extend with new chains