OpenMind/OM1 #361: [Bounty] integrate 3js simulator

Prachi1615, 2025-08-29:
**Overview**
OM1 already supports Gazebo for advanced robotics simulation. However, Gazebo requires ROS and a heavy setup. To make OM1 more accessible, this bounty challenges contributors to build a browser-based three.js simulator that allows developers to run OM1 without installing ROS or having real hardware.

**Features**

- Web Simulator: Build a simple robot simulator in three.js that runs in the browser.
- OM1 Control: Robots should accept OM1 commands (move, rotate, lift, stop) through the existing API layer.
- Environment: Simple playground (floor, walls, or obstacles).

**Guidelines**

- Keep integration modular 
- Provide starter robot templates
- Document how to run the sim in a browser with OM1
- Provide a demo video of OM1 controlling the simulator
- Prioritize accessibility 

For submission, comment on the issue with the following details
- 🤖 Robot(s) Simulated: [List types you added]  
- 🚀 PR Link: [Link to the PR]  
- 🎥 Demo Video: [YouTube/Vimeo/Drive URL]  
- 📑 Notes: [Setup steps / limitations]

**License**
This project is licensed under the MIT License. See the [LICENSE](https://github.com/OpenmindAGI/OM1/blob/main/LICENSE) file for more details.

**Contact**
For questions or further assistance, reach out via the issue tracker or contact the maintainers directly.

debaa98, 2025-08-30:
is there any specific bounty amount?

Wanbogang, 2025-10-02:
Hi @Prachi1615 — could you please review my submission?

**Robot(s) Simulated:** Differential-drive wheeled bot + simple playground (floor, walls, obstacles)

**PR Link:** https://github.com/OpenMind/OM1/pull/420

**Demo Video:** https://youtu.be/Wrb7SLFoIoI

**Notes (setup & limitations):**
- Bridge (Express + WS):
  cd examples/threejs-sim/bridge
  node index.js   # serves at 0.0.0.0:8081
- Client (React + Three.js):
  cd examples/threejs-sim/client
  npm i
  echo "VITE_BRIDGE_WS=ws://localhost:8081/ws" > .env
  npm run dev -- --host 0.0.0.0 --port 5173
- HUD: pose, beams (sensors), minDist, reward; manual WASD control.
- Loop is OM1-style compatible; physics simplified (no contact/IK).
  
Thanks!

mudaly10-coder, 2025-10-19:
Hi! I can take this on and deliver a fix/PR.
If you prefer a different approach or scope, let me know and I’ll adjust.

lau90eth, 2025-11-18:
Bounty #361 completed

🤖 Robot(s) Simulated: N/A (voice input plugin – works with any robot)

🚀 PR Link: https://github.com/lau90eth/OM1/pull/3

🎥 Demo Video: https://youtu.be/SbvrJ0D18Ws (20 sec, 2–3 speakers + beamforming)

📑 Notes:
- Real multi-speaker detection (2–3 voices)
- Beamforming simulation (delay-and-sum, dominant angle)
- Mock mode enabled (no hardware required)
- Hardware-ready (PyAudio + microphone array)
- Test suite 100% passed
- Configurable via src/plugins/voice/config.yaml
- Ready for any robot (just plug the plugin)

@Prachi1615 ready for review/merge – thanks!

ismoilh, 2026-01-08:
I've submitted a PR for this bounty:
   
   🤖 **Robot(s) Simulated**: Basic wheeled robot (box robot with 4 wheels)
   🚀 **PR Link**: https://github.com/OpenMind/OM1/pull/1124
   🎥 **Demo Video**: https://www.youtube.com/watch?v=uIKtt-yMW7E
   📑 **Notes**: 
   - Setup: `uv run src/run.py web_sim` → open `http://localhost:8001`
   - No ROS or hardware required
   - Documentation files were generated with AI assistance

cc: @Prachi1615