OpenMind/OM1 #362: [Bounty] Integrate a MuJoCo Simulator

Prachi1615, 2025-08-29:
**Overview**
OM1 currently supports Gazebo for robotics simulation. While Gazebo is great for robotics and ROS workflows, MuJoCo offers high-performance physics and is widely used in robotics control and reinforcement learning.
This bounty challenges contributors to integrate MuJoCo with OM1, enabling developers to run realistic robot control through OM1.

**Features**

- MuJoCo Environment: Set up a basic MuJoCo environment with a robot model (e.g., wheeled bot, robotic arm).
- OM1 Control Bridge: Commands from OM1 should control the MuJoCo robot in real-time (move, rotate, lift, etc.).
- Sensor Feedback: Stream simulated sensor data (joint states, positions, forces) back into OM1
- Demo Scripts: Provide examples showing how OM1 can run a robot simulation in MuJoCo.
- Config Switch: Allow users to select between Gazebo, MuJoCo, or three.js sim for flexibility.

**Guidelines**

- Keep the integration modular.
- Use MuJoCo’s Python API (or bindings) for control and data flow.
- Provide robot XML models (MJCF/URDF) for at least one simple robot.
- Document installation requirements (MuJoCo license, setup steps). 
- Raise a separate PR to update existing docs with installation details.
- Provide a short demo video of OM1 controlling a MuJoCo robot.

For submission, comment on this issue with
- 🤖 Robot(s) Simulated: [List the robot you integrated]  
- 🚀 PR Link: [Link to the PR you raised] , [Link to documentation PR]
- 🎥 Demo Video: [YouTube/Vimeo/Drive URL]  
- 📑 Notes: [Setup steps, limitations, improvements]

🤖 Robot(s) Simulated: [List types you added]
🚀 PR Link: [Link to the PR]
🎥 Demo Video: [YouTube/Vimeo/Drive URL]
📑 Notes: [Setup steps / limitations]

License
This project is licensed under the MIT License. See the [LICENSE](https://github.com/OpenmindAGI/OM1/blob/main/LICENSE) file for more details.

Contact
For questions or further assistance, reach out via the issue tracker or contact the maintainers directly.

Wanbogang, 2025-10-02:
**Submission: OM1 MuJoCo Bridge + HTTP Adapter (Minimal Demo)**

**Repo:** https://github.com/Wanbogang/om1-mujoco-bridge  
**Tag/Release:** v0.1.0

**What’s included**
- `om1_bridge/bridge.py` — MuJoCo stepper with WebSocket API (`reset`, `set`, `step`, `state`)
- `om1_bridge/adapter.py` — HTTP adapter exposing `/control` and forwarding to the WS bridge
- `run_all.sh` — runs both bridge + adapter together (tmux-friendly)
- `assets/simple_arm.xml` — simple 2-DOF arm model (stable params)

**How to run**
```bash
./run_all.sh
```
Send a control command
```
curl -s -X POST http://127.0.0.1:8088/control \
  -H 'Content-Type: application/json' \
  -d '{"j1": 25, "j2": -35, "steps": 250}' | jq .
```
Expected response
```
{
  "ok": true,
  "result": {
    "ok": true,
    "obs": {
      "qpos": [...],
      "qvel": [...],
      "time": ...
    }
  }
}
```
**Requirements**
- Python 3.10+
- Virtualenv
- Packages: `mujoco`, `numpy`, `websockets` (see `requirements.txt`)

**Release**
- https://github.com/Wanbogang/om1-mujoco-bridge/releases/tag/v0.1.0

**Notes**
- Headless MuJoCo; joint damping + moderate `kp` for stability.
- Runs cleanly inside `tmux`; easy to detach/reattach for demos.

lau90eth, 2025-11-13:
**BOUNTY #362 COMPLETATO!**

**Repo**: https://github.com/lau90eth/mujoco-openmind-bounty

**File**:
- `pendulum.xml`
- `simulate_headless.py`
- `gym_env.py`
- `README.md`

**Test**:
```bash
python simulate_headless.py

ScapeXYZ, 2025-12-06:
## Bounty Submission: MuJoCo Simulator Integration (Bounty #362)
This submission completes the OM1 Control Bridge for the MuJoCo simulator, specifically integrating the Inverted Pendulum system. The implementation allows external OM1 actions to successfully send and apply force commands to the physics environment in real-time.

Pull Request (PR) Link	https://github.com/OpenMind/OM1/pull/814
Demo Video Link	https://youtu.be/DnkBuWaNkH4
Robot Simulated	Inverted Pendulum (Cart-Pole System)

## Setup & Execution Steps
The setup requires two active terminals to demonstrate the control loop:

**Terminal 1 (Server/Adapter)**: Navigate to simulators/mujoco and launch the adapter:
`python om1_mujoco_api_server.py`
The server initializes MuJoCo and then stays silent, as sensor stream errors are suppressed for a clean demo.
**Terminal 2 (Client Action)**: Execute the control command. This simulates the OM1 system sending a force of 50.0 N to the simulator:
`Invoke-WebRequest -Method Post -Uri http://127.0.0.1:8080/control -ContentType "application/json" -Body '{"force": 50.0}'`
**Verification**: As shown in the video, running the command in Terminal 2 immediately triggers a clear log in Terminal 1: Received control command: `set force to 50.0 N`.

dominitas, 2025-12-27:
Hi @Prachi1615   ,

I noticed this issue. We at Roxonn help teams and developers pay for completed GitHub tasks without hiring full-time contractors.

If helpful, I’d be happy to show how Roxonn can get this task done quickly.

– Domantas

Prachi1615, 2026-01-19:
Hi everyone, thanks for working on this issue.
We've decided this feature is now out of scope of the project, hence closing it now.

closed 2026-01-19 by Prachi1615