OpenMind/OM1 #363: [Bounty] Improve the existing Gazebo simulator environment

Prachi1615, 2025-08-29:
**Overview**
OM1 already integrates with Gazebo to simulate robots. However, the current environment is fairly minimal. This task focuses on enhancing the existing Gazebo world, robot models, and interaction so developers can run richer OM1 workflows and tests.

**Features**

- Environment Enhancements: Add obstacles, furniture, or test arenas for robots.
- Data Feedback: Stream richer sensor data (vision, range, IMU, etc.) back into OM1.
- Demo Script: Example scenario (e.g., robot navigating a room, reporting data).

**Guidelines**

- Keep improvements modular (new worlds/models should not break existing setups).
- Ensure new models follow ROS/Gazebo best practices.
- Provide documentation updates with instructions if necessary.
- Include a demo video showing OM1 controlling a robot in the improved Gazebo world.
- Test on common setups (Ubuntu + ROS + Gazebo).

For submission, comment on the issue with
- 🌍 Improvements Made: [Environment/Models/Devices added]  
- 🚀 PR Link: [Link to your PR]  
- 🎥 Demo Video: [YouTube/Vimeo/Drive URL]  
- 📑 Notes: [Setup steps, limitations, future ideas]

**License**
This project is licensed under the MIT License. See the [LICENSE](https://github.com/OpenmindAGI/OM1/blob/main/LICENSE) file for more details.

**Contact**
For questions or further assistance, reach out via the issue tracker or contact the maintainers directly.

Wanbogang, 2025-10-02:
Hi team 👋

I’d like to take on bounty #363. For the first milestone, my focus will be:
- Reproducible setup (scripts + quickstart docs)
- Deterministic world/seed
- Basic CI smoke test
- Short demo (video) and NOTES.md

Plan:
1. Fork the repo and create a dedicated branch for this bounty.
2. Open a draft PR with initial scaffolding (scripts + docs placeholders) so progress is visible.
3. Update the PR step by step until Milestone-1 is complete, then add the unlisted demo video link.

I’ll share the draft PR link here soon.  
Thanks!

Wanbogang, 2025-10-02:
Draft PR for visibility:\n\n- \n\nI'll keep updating with code, docs, and the unlisted demo video link.

lau90eth, 2025-11-13:
**BOUNTY #363 COMPLETATO!**

**Repo**: https://github.com/lau90eth/gazebo-bounty-363

**File**:
- `go2_improved.sdf`: Robot Go2 con **LiDAR 3D + IMU realistico**
- `improved_test.world`
- `launch/improved_gazebo.launch.py`
- `README.md`

**Test**:
```bash
gazebo improved_test.world
ros2 topic echo /scan
ros2 topic echo /imu

closed 2025-12-19 by Prachi1615