OpenMind/OM1 #365: [Bounty] New input plugins for integrating new sensors

Prachi1615, 2025-08-29:
**Overview**
Currently OM1 supports certain sensory inputs. We want OM1 to support more sensor inputs like temperature, humidity(or any other).

**Features**

- New Sensor Drivers - Implementation of input plugins for the new sensors (e.g., temperature, humidity, light, air quality, etc.).
- Real-time data reading from the sensors.
- Ability for OM1 to access and utilize new sensor data seamlessly.
- Ensure minimal latency and efficient resource usage.
- Test cases for sensor data accuracy and stability.
- Demonstration of integration with OM1’s existing system features.
- Documentation for configuration, and usage.

**Guidelines**

- Keep your implementation modular.
- Provide clear documentation.
- Provide a demo video.

For submission, comment on the issue with following details

🤖 Sensor(s) Integrated: [e.g., Humidity, Temperature, etc]
🚀 PR Link: [Link to your PR], [Link to documentation PR]
🎥 Demo Video: [YouTube/Drive URL]
📑 Notes: [Setup steps, limitations, improvements]

**License**
This project is licensed under the MIT License. See the [LICENSE](https://github.com/OpenmindAGI/OM1/blob/main/LICENSE) file for more details.

**Contact**
For questions or further assistance, reach out via the issue tracker or contact the maintainers directly.

aybanda, 2025-08-30:
Hi! @Prachi1615 I'm interested in this bounty.
Is this open for everyone to work on? Could you please provide the bounty details and any specific requirements?
Thanks!

Prachi1615, 2025-09-01:
@aybanda 
It is open for everyone to work on. Make sure you read the Wiki attached for reference in the description. All the details are available there.

Wanbogang, 2025-09-29:
I’ll implement an input plugin for **BME280** (temp/humidity/pressure).
Why BME280: easy to source & stable baseline. Plan: simple polling loop,
env/yaml config, mock mode (no hardware), and a short demo video.
I’ll open a PR soon. —Wanbogang

sohw400, 2025-10-17:
### Submission for Issue #365

🤖 **Sensor(s) Integrated**: DHT22 (Temperature & Humidity), BH1750 (Ambient Light)

🚀 **PR Links**: 
- DHT22: https://github.com/OpenMind/OM1/pull/459
- BH1750: https://github.com/OpenMind/OM1/pull/460

🎥 **Demo Video**: Testing was done in mock mode with automated test suites. Both sensors include mock mode that generates realistic data, making them testable without physical hardware.

📑 **Notes**:

**DHT22 (Temperature & Humidity Sensor)**
- Reads temperature in Celsius/Fahrenheit and relative humidity
- Supports both hardware mode (Raspberry Pi GPIO) and mock mode
- Mock mode generates realistic indoor conditions (20-24°C, 50-60% humidity)
- Includes comfort level interpretation (cool, comfortable, warm)
- Full setup guide in `docs/sensors/DHT22_SETUP.md`
- Test coverage: 26 unit tests

**BH1750 (Ambient Light Sensor)**
- Measures illuminance from 1-65535 lux via I2C
- Supports multiple I2C addresses (0x23, 0x5C)
- Mock mode simulates typical indoor lighting (250 lux with variations)
- Provides contextual descriptions (dark, dim, moderate, bright, very bright)
- Full setup guide in `docs/sensors/BH1750_SETUP.md`
- Test coverage: 29 unit tests

**Setup**: Both sensors work with `mock_mode: true` for development without hardware. For production use, see the respective setup docs for wiring diagrams and dependencies.

**Improvements**: 
- Automatic fallback to mock mode when hardware libraries unavailable
- Thread-safe sensor polling
- Comprehensive error handling
- Natural language output for LLM consumption

**Limitations**: Hardware testing was done on Raspberry Pi. Other platforms should work but may need library adjustments.

lau90eth, 2025-11-17:
Bounty #365 completed

PR: https://github.com/lau90eth/OM1/pull/1
Demo Video: https://youtu.be/SbvrJ0D18Ws
Contributor: @lau90eth

Implemented 4 sensor plugins:
• BME280 (temperature, humidity, pressure + comfort level)
• DHT22 (temperature, humidity + comfort)
• BH1750 (ambient light + natural language description)
• MQ-135 (CO₂ ppm + air quality classification)

Features:
- Mock mode enabled by default (works without hardware)
- Hardware-ready (just set mock_mode: false)
- 100% test coverage
- Real-time data with natural language output

Happy to make any adjustments if needed. 
Thank you @Prachi1615

ScapeXYZ, 2025-11-28:
Sensors Integrated: 
Temperature
 Humidity
Light
Air Quality (CO₂)
Sound
 PR Link: https://github.com/OpenMind/OM1/pull/770
 Demo Video: https://youtu.be/R_VwnA8bhLk

 Notes:

- Sensors run independently with live mock-mode updates (no hardware required).
- Values refresh continuously and expose a simple get_data() interface for OM1 integration.
- Test file included to validate each sensor’s return fields , all tests pass successfully.

hyaku0121, 2025-11-28:
🤖 Sensor(s) Integrated: 
**System Health Monitor** (Internal Telemetry)
- CPU Usage (%)
- RAM Usage (%)
- Disk Usage (%)
- Battery Level & Charging Status

🚀 PR Link: https://github.com/OpenMind/OM1/pull/771

🎥 Demo Video: https://www.youtube.com/watch?v=SX16nmOdEAU

📑 Notes:
- **Real-time Reaction:** The demo video shows the sensor accurately tracking CPU load spikes (0% -> 100% -> 0%) in real-time using a stress test script.
- **Implementation:** Uses `psutil` for cross-platform support with zero external hardware dependency.
- **Quality Assurance:** Unit tests included (`tests/inputs/test_system_health.py`) covering initialization and data polling logic.
- **Verification:** Verified with `config/test_health.json5`.

ScapeXYZ, 2025-12-05:
Bounty https://github.com/OpenMind/OM1/issues/365 completed

Sensor(s) Integrated: Temperature, Humidity, Light Intensity, Air Quality (CO₂), Sound.

 PR Link: [INSERT YOUR PR LINK HERE]

Demo Video: https://youtu.be/ZT01ztqYUSY

📑 Notes: 
- **Implementation:** Plugins are modularized and placed in the correct OM1 structure (`src/inputs/env_sensors/`). Configuration is provided in `config/multisensor_demo.json5`.
- **Testing:** Includes Pytest verification (`tests/inputs/test_multisensors.py`).
- **Limitation Justification:** The demo video was generated using a temporary standalone script (`demo_run.py`) instead of the official `src/run.py`. This was necessary due to persistent, non-bounty-related dependency conflicts (cortex_llm/TensorFlow wheels) on the local Windows environment. The submitted code is fully compatible with the OM1 runtime structure.

Farukest, 2026-01-03:
🤖 Sensor(s) Integrated: AWAIR Element (Temperature, Humidity, CO2, VOC, PM2.5)

🚀 PR Link: https://github.com/OpenmindAGI/OM1/pull/949

🎥 Demo Video: https://youtu.be/X1jDjoVPg9c

📑 Notes:
- Local API & Cloud API support
- Real-time polling with configurable interval
- Includes working LG ThinQ AC control demo (LLM decides heating/cooling based on sensor data)
- Full setup guide in docs/awair_integration.md
- Unit tests included

dolepee, 2026-02-01:
🤖 **Sensor(s) Integrated:** Temperature, Humidity, Light, Air Quality

🚀 **PR Link:** https://github.com/OpenmindAGI/OM1/pull/2052

📑 **Notes:**

**Implementation Details:**
- **Temperature Sensor** (DHT22_SIM): Simulated readings with realistic variations
- **Humidity Sensor** (DHT22): Hardware support for DHT22 sensor with I2C
- **Light Sensor** (BH1750): Hardware support for BH1750 ambient light sensor
- **Air Quality Sensor** (AIR_QUALITY_SIM): Simulated CO2, VOC, PM2.5 monitoring

**Setup:**
All sensors follow OM1's FuserInput plugin architecture. Configuration examples:
```json5
{
  type: "TemperatureSensor",
  sensor_type: "simulation",
  update_interval: 2.0,
  calibration_offset: 0.0
}
```

**Limitations:**
- Hardware sensors require physical I2C devices (DHT22, BH1750)
- Simulation modes provided for testing without hardware

**Improvements:**
- Modular design allows easy addition of more sensors
- Calibration offset support for sensor tuning
- Async implementation for efficient resource usage