OpenMind/OM1 #366: [Bounty] Smart device communication

Prachi1615, 2025-08-29:
**Overview**
Integrate OM1 with Home Assistant to control a basic IoT device (e.g., smart light, thermostat, or switch). The solution should allow users to turn devices on/off, adjust settings.

**Features**

- Enable OM1 to communicate with Home Assistant via supported protocols (e.g., REST API, MQTT, WebSocket).
- Support basic control operations:
  - On/Off for switches and lights
  - Brightness/Color adjustment for lights
  - Temperature setting for thermostats

**Guidelines**
Keep your implementation modular.
Provide clear documentation.
Provide a demo video.
For submission, comment on the issue with following details

🤖 Home Assistant Integrated: [Alexa, Siri]
🚀 PR Link: [Link to your PR] 
🎥 Demo Video: [YouTube/Drive URL]
📑 Notes: [Setup steps, limitations, improvements]

**License**
This project is licensed under the MIT License. See the [LICENSE](https://github.com/OpenmindAGI/OM1/blob/main/LICENSE) file for more details.

**Contact**
For questions or further assistance, reach out via the issue tracker or contact the maintainers directly.

Asriadi13, 2025-09-06:
Hello, I’d like to work on this bounty. May I take this issue?

Prachi1615, 2025-09-24:
Noted. Will update.
@Shaurya-M002

mudaly10-coder, 2025-10-19:
Auto-claim: I'm taking this on.
- Part of an automated bounty triage/claim workflow.
- If you'd like a different approach or more detail, please say so.

Prachi1615, 2025-10-20:
@meekMusa2 
Can you share the link to your fork where you have implemented this?
or if you can share you json config too it'll be helpful to debug.

Prachi1615, 2025-10-20:
@Wanbogang 
Reviewing your PRs

lau90eth, 2025-11-18:
Bounty #366 completed

🤖 Home Assistant Integrated: Alexa, Siri (multi-voice input with beamforming)

🚀 PR Link: https://github.com/lau90eth/OM1/pull/4
🎥 Demo Video: https://www.youtube.com/watch?v=-S75yZflElc (20 sec, 2–3 speakers + beamforming)

📑 Notes:
- Real multi-speaker detection (2–3 voices)
- Beamforming simulation (delay-and-sum, dominant angle)
- Mock mode enabled (no hardware required)
- Hardware-ready (PyAudio + microphone array)
- Test suite 100% passed
- Configurable via src/plugins/voice/config.yaml

@Prachi1615 ready for review/merge – thanks!

lau90eth, 2025-11-28:
Bounty #366 completed

🏠 Philips Hue Integration: Full smart home control
🚀 PR Link: OpenMind/OM1#775

📑 Features:
- Light control (on/off, brightness, color, color temperature)
- Group management
- Scene activation
- Bridge discovery and authentication
- Complete error handling

Test suite included. @Prachi1615 ready for review – thanks!

dominitas, 2025-12-27:
Hi @Prachi1615 @meekMusa2 @ScapeXYZ @lau90eth @Ore0luwa   ,

I noticed this issue. We at Roxonn help teams and developers pay for completed GitHub tasks without hiring full-time contractors.

If helpful, I’d be happy to show how Roxonn can get this task done quickly.

– Domantas

Wanbogang, 2026-03-01:
Hi @Prachi1615 
If you have free time I would like you to check it out and give me feedback #2435 
Thanks