OpenMind/OM1 #367: [Bounty] Smart device communication

Prachi1615, 2025-08-30:
**Overview**
Enable OM1 to integrate with a smart assistant (e.g., Home Assistant) to help a user order something, with payment processed through a crypto wallet (e.g., Coinbase).

**Features**

- OM1 communicates with the assistant to place orders or trigger actions.
- Wallet-Based Payments 
  - Payments are executed securely via a connected crypto wallet.
  - Transaction status (success, pending, failed) is reported back to the user.
- Demonstrate a full workflow: user requests order → OM1 communicates with assistant → payment processed via wallet → confirmation received.

**Guidelines**

  - Keep your implementation modular.
  - Provide clear documentation.
  - Provide a demo video.
  - For submission, comment on the issue with following details

🤖 Home Assistant Integrated: [e.g., Alexa, Siri]
🚀 PR Link: [Link to your PR] 
🎥 Demo Video: [YouTube/Drive URL]
📑 Notes: [Setup steps, limitations, improvements]

**License**
This project is licensed under the MIT License. See the [LICENSE](https://github.com/OpenmindAGI/OM1/blob/main/LICENSE) file for more details.

**Contact**
For questions or further assistance, reach out via the issue tracker or contact the maintainers directly.

PeterWanjihia, 2025-09-01:
Hello @Prachi1615  can I work on thi issue ??

Prachi1615, 2025-09-01:
@PeterWanjihia 
Yes, you can start working on this issue. Make sure you read the Wiki attached for reference in the description.

meichbao, 2025-09-19:
还不能参加测试呢

Wanbogang, 2025-10-21:
Home Assistant used: Yes — home_assistant_config.yaml

Repo: https://github.com/Wanbogang/om1-emoji-order-assistant  
PR: https://github.com/Wanbogang/om1-emoji-order-assistant/pull/2

Demo Video: https://youtu.be/eB4q9EvyHDA  

Notes:
- Run: ./demo/e2e_wallet_demo.sh
- Open: http://localhost:8080/ (browser with MetaMask on testnet/local)
- Flow shown: voice → OM1 creates order → user signs via MetaMask → server records txHash (verify with `curl http://localhost:8080/api/orders`)

Artifacts in PR:  
- web-client/ (demo wallet UI)  
- server.py (FastAPI demo + /api/payment/signed + /api/orders)  
- demo/e2e_wallet_demo.sh  
- home_assistant_config.yaml  
- README.md (Claim bounty section updated)

Thank you — looking forward to verification.

timhaldorsson, 2025-10-24:
it this bounty still ongoing? or was the wallet payments sorted.

meekMusa2, 2025-11-19:
🎉 Bounty Submission - Charity Donations with Cryptocurrency Payments
🤖 Home Assistant Integrated
Google Assistant (via Home Assistant)
🚀 PR: https://github.com/OpenMind/OM1/pull/735
🎥 Demo Video: https://youtu.be/ZHSAFOm7NM4
🚀 Post Link: https://x.com/MusaAbdulquadr2/status/1990994546855956634?s=20
📑 Implementation Details
Overview:
Implemented a complete charity donation system where users can donate to verified causes using cryptocurrency through voice commands, with real blockchain transactions and wallet integration.

ScapeXYZ, 2025-11-26:
Bounty Submission - Pizza Order with Cryptocurrency Payments
Home Assistant Integrated - Siri
PR:  https://github.com/OpenMind/OM1/pull/765
Demo Video : https://youtu.be/WheMjXKIY48
Post Link: https://x.com/temitope_isola8/status/1993677748334707062?s=20

kvzuobai, 2026-01-06:
Hi @Prachi1615,

I'm interested in working on this bounty (#367). I have experience with blockchain development, crypto wallet integrations (Coinbase, Web3.py), and building scalable applications.

Plan:
- Integrate OM1 with Home Assistant (Alexa/Siri)
- Implement secure crypto wallet payment processing
- Create a modular plugin architecture
- Provide comprehensive documentation and demo video
- Full test coverage

I've filled out the credits form and joined the Discord/Telegram community. Looking forward to discussing the technical details.

Thanks!

kvzuobai, 2026-01-06:
## 🎉 Bounty Submission - Smart Assistant + Wallet Payments Integration

### 🤖 Home Assistant Integrated
- Home Assistant REST API
- Amazon Alexa Skills support
- Siri Shortcuts compatible

### 🚀 PR Details
[#1017 - feat: Smart Assistant + Wallet Payments Integration (Bounty #367)](https://github.com/OpenMind/OM1/pull/1017)

### 📑 Implementation Details

**Features Implemented:**
- ✅ Smart assistant integration (Home Assistant, Alexa)
- ✅ Crypto wallet payment processing (Coinbase, Mock)
- ✅ Multi-network support (Ethereum, Polygon, Base, Solana)
- ✅ Real-time transaction status tracking
- ✅ Modular, extensible architecture
- ✅ Comprehensive documentation & error handling

**Module Structure:**
```
src/plugins/smart_assistant_wallet/
├── __init__.py                 # Module exports  
├── assistant_connector.py       # Smart assistant integrations
├── wallet_handler.py            # Wallet payment handling
└── README_BOUNTY_367.md         # Complete documentation
```

### 📋 Submission Checklist
- [x] Code complies with contribution guidelines
- [x] Full type hints (PEP 484)
- [x] Comprehensive logging
- [x] Modular architecture with factory patterns
- [x] Detailed README with examples
- [x] Mock wallet for testing
- [x] Real wallet integration ready (Coinbase)

### 🔗 Links
- **GitHub PR**: [#1017](https://github.com/OpenMind/OM1/pull/1017)
- **Repository**: [kvzuobai/OM1](https://github.com/kvzuobai/OM1)

Ready for review! 🚀

Himess, 2026-01-08:
🤖 **Home Assistant Integrated**: HTTP Webhook (compatible with Alexa, Siri, Home Assistant, and any smart assistant that can make HTTP requests)

🚀 **Post Link**: [Coming soon - will update]

🎥 **Demo Video**: https://youtu.be/RwkGhxWCRqc

📑 **Notes**:
- **Setup**: Requires Ollama (or OpenAI API) + Coinbase CDP credentials
- **Input Plugin**: HTTP webhook server on configurable port (default: 8080)
- **Action Plugin**: Coinbase CDP wallet connector for crypto payments
- **Supported Commands**: `pay 0.001 ETH to 0x...` or `send 0.5 USDC to 0x...`
- **Network**: Base Sepolia testnet (configurable)
- **Tests**: 30 unit tests (13 for input, 17 for action)
- **Real Transactions**: Successfully executed on Base Sepolia

**PR**: #1133

steadii, 2026-01-08:
🤖 Home Assistant Integrated: Mock Home Assistant (Home Assistant–compatible interface)

🚀 Post Link: https://x.com/YusufIbrahimAb7/status/2009354465111928894?s=20

🎥 Demo Video: https://youtu.be/xEfSFnzNNFg

📑 Notes:
- Environment: WSL Ubuntu
- Implemented a full end-to-end workflow: user request → OM1 → smart assistant → wallet payment → confirmation
- Smart assistant layer is mocked for lightweight setup but follows Home Assistant–compatible patterns
- Wallet payment is simulated to demonstrate transaction execution and status reporting

steadii, 2026-01-08:
## Overview
This pull request submits a bounty contribution demonstrating an end-to-end OM1 workflow that integrates a smart assistant interface with wallet-based payments.

The implementation shows how a user request can be handled by OM1, routed through a smart assistant to place an order, processed via a crypto wallet payment, and confirmed back to the user.

(linked issue: https://github.com/OpenMind/OM1/issues/367)

## Type of change
- [ ] Bug fix
- [ ] New feature
- [ ] Documentation improvement
- [x] Bounty issue submission
- [ ] Other:

## Changes
- Added a modular smart assistant integration compatible with Home Assistant–style workflows
- Implemented a wallet payment flow with transaction status reporting
- Created an end-to-end workflow controller connecting assistant actions and wallet payments
- Added a runnable demo script to demonstrate the full order → payment → confirmation lifecycle
- Updated documentation to explain setup, architecture, and usage
- Ensured all changes are isolated from OM1 core runtime files

## Checklist
- [x] Code complies with style guidelines
- [x] Self-review completed
- [x] Documentation updated (if applicable)
- [x] Local testing completed
- [ ] Tests added/updated (if applicable)

## Impact
This contribution provides a practical reference implementation for assistant-driven commerce workflows within OM1. It introduces no breaking changes, does not modify core runtime logic, and keeps all new functionality modular. The demo uses mocked components for ease of testing while remaining compatible with real smart assistant and wallet integrations.

## Additional Information
- Environment: WSL Ubuntu
- Smart assistant layer is mocked for a lightweight demo but follows Home Assistant–compatible request/response patterns
- Wallet payments are simulated to safely demonstrate transaction execution and status reporting
- Demo script output confirms successful order placement and payment completion

Prachi1615, 2026-01-19:
Hi everyone, thanks for working on this issue.
We've decided this feature is now out of scope of the project, hence closing it now.