OpenMind/OM1 #404: Confusion during first run with uv run src/run.py spot

imsardine, 2025-09-22:
Following the instructions in the README, when I first ran uv run src/run.py spot, the logs showed:

> POST https://api.openmind.org/api/core/openai/chat/completion ... Insufficient balance. Please add funds to your account at https://portal.openmind.org

It looks like the project accesses the OpenAI API indirectly through https://api.openmind.org/, which seems to bind it to OpenMind’s service. To test this, I purchased $10 in credits.

After running the command again, it started using the camera, but the logs kept repeating the same pattern. When I opened http://localhost:8000 (WebSim), I wasn’t sure what I was supposed to observe.

[image]

I’m not sure if I did something wrong, but while following the “Getting Started” steps, I felt lost about the actual goal. I was hoping to see some meaningful result once it worked, but it seemed like nothing was happening.

It would be really helpful if the documentation could provide more guidance on what to expect and how to verify that everything is running correctly. Thanks!

openminddev, 2025-09-22:
Hi @imsardine , thank you for your feedback. @Prachi1615 Could you please provide more insights here? Also, we would love to give me more free credits. Please email your account address to boyuan@openmind.org.

Prachi1615, 2025-10-02:
thanks. will update that.

After you run the agent, you can try talking to the agent and check the logs as well.

Prachi1615, 2025-10-05:
Hi @imsardine 
I hope your issue is resolved now.
Closing the issue now.

closed 2025-10-05 by Prachi1615