OpenMind/OM1 #417: ASR WS: need exact payload schema; always "Audio Timeout" or "'audio' field missing"

Wanbogang, 2025-09-29:
Description
**Describe the bug**
I can connect to `wss://api.openmind.org/api/core/google/asr`, but the server never accepts the audio:
- If any frame lacks `audio` → `Invalid message format: 'audio' field missing`
- If *every* frame includes `audio` (base64 PCM LINEAR16) → `Transcription error: 400 Audio Timeout Error: Long duration elapsed without audio.`

This suggests the gateway requires a specific WS payload schema (field name, ordering, end-of-stream, or binary vs JSON) that is not documented on our side.

**To Reproduce**
1. Prepare a WAV 16 kHz / mono / 16-bit file at `tools/asr-eval/en/001.wav`.
2. Use Python 3.10 + `websockets==15.0.1`.
3. Run the minimal repro below (API key redacted):

```python
import os, json, base64, asyncio, wave, websockets

URL = "wss://api.openmind.org/api/core/google/asr?api_key=****"
WAV = "tools/asr-eval/en/001.wav"

def read_pcm(path):
    with wave.open(path, "rb") as w:
        assert (w.getframerate(), w.getnchannels(), w.getsampwidth()*8) == (16000, 1, 16)
        return w.readframes(w.getnframes()), 16000

async def main():
    pcm, sr = read_pcm(WAV)
    payload = {
        "config": {
            "languageCode": "en-US",
            "encoding": "LINEAR16",
            "sampleRateHertz": sr,
            "channels": 1,
            "bitsPerSample": 16
        },
        "audio": base64.b64encode(pcm).decode("ascii")
    }
    async with websockets.connect(URL, max_size=None) as ws:
        await ws.send(json.dumps(payload))
        try:
            while True:
                print("[<-]", await asyncio.wait_for(ws.recv(), timeout=10))
        except asyncio.TimeoutError:
            pass

asyncio.run(main())
```

Actual behavior (sample replies)
```
{"type":"connection","message":"Connected to ASR service","clientId":"<example-id>"}
{"type":"error","message":"Transcription error: 400 Audio Timeout Error: Long duration elapsed without audio."}
```
```
**Recent clientIds seen in replies (for server-side correlation):**
- `9eb3e6e7-c274-4ef7-a36c-5d98f0e1cf4a`
- `205434b0-5b4b-4db1-b3f1-5ccffb90ecc2`
- `e382c8e9-c7bd-4c5f-93de-3af6c2ade635`
- `b2933f95-0d1b-4eb5-a2d4-22fd47c1cb9b`
```

[image]
[image]

If a message is sent without audio, the server returns:
```
{"type":"error","message":"Invalid message format: 'audio' field missing"}
```

Expected behavior
A minimal working example of the first WS message that your official client sends (and how subsequent + final frames must look), so the server actually processes audio and returns transcript results (e.g., result(s), final, transcript, etc.).

Screenshots

-   OM1 portal (credits & API keys) – API key redacted
-   Console logs showing the responses above
(I will attach the screenshots.)

System Information

-   OS: Ubuntu Linux (server)
-   Python: 3.10
-   websockets: 15.0.1
-   Audio: WAV 16 kHz / mono / 16-bit (LINEAR16)

Additional context
Already tried:

-   JSON streaming where every frame contains audio (no sleeps), with key variants: audio, chunk, data, audioContent, audio_base64
-   Root vs typed frames: {"start":true,"audio":...}, {"type":"start" | "audio" | "end"}, {"end":true}/{"stop":true}
-   Binary PCM frames after a JSON config hello
-   Subprotocols: none, om1-asr, asr, google-asr, speech, asr.v1, om1.v1, v1, json
All connect successfully but end in the same timeout.

What we need to proceed
Please share the exact payload/handshake the server expects:

1.Frame format: text JSON or binary?
2.Correct audio field name and type: base64 string or raw bytes?
3.Required root fields: requestId, starting seq (0/1), mode, end/stop, etc.
4.Required Sec-WebSocket-Protocol value (if any)?
5.One or two valid follow-up frames and the final end-of-stream frame.

Once I have this, I can align our client 1:1 and (if helpful) submit a PR to improve the evaluator/docs.

Acopper1, 2025-10-01:
This help mate thank for making community work simpler

closed 2025-10-21 by Prachi1615

Prachi1615, 2025-10-21:
@Wanbogang 
Closing this issue as a duplicate issue exists.
Track below linked issue for progress 
https://github.com/OpenMind/OM1/issues/406

Wanbogang, 2025-10-21:
@Prachi1615 
Thank You